miblooming/JZ-TS24-2 #1: Support russian language?!?

jhnf, 2017-04-18:
Hi, great display! I'd like to see the Russian language on your display, is it possible ?. Anyway, I can help with the translation.

jhnf, 2017-04-19:
Hello!
I figured out how to translate the ui_editor, but the font is ugly for the Russian language. It would be nice to change the font to another one, like Cambria.

miblooming, 2017-04-23:
Yes, I have send you email.

Muhin555, 2017-09-01:
I do not know the Chinese language. So I can't buy a display Taobao. There is a possibility to buy it on AliExpress? A lot of my friends in Russia would like to buy it.

miblooming, 2017-09-01:
Yes, I'll upload this product to AliExpress these days. Then send  the URL link to you.

miblooming, 2017-09-01:
@Muhin555  Paypal is ok for order. Upload product to AliExpress it too expensive. Email to me for detail, thank you. My email is loomingxu@qq.com

Muhin555, 2017-09-06:
I'm send letter to email loomingxu@qq.com

miblooming, 2017-09-06:
@Muhin555 Your email is classified as a spam email, but I find it now. And i have reply your email.

Muhin555, 2017-09-06:
Ок. I'm read your email.

miblooming, 2019-05-15:
Russian is supported now.
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5)

closed 2019-05-15 by miblooming