miblooming/JZ-TS24-2 #3: Not display temperature

DaGov, 2017-05-16:
Not display temperature T0 extruder, but heated on. T1 and bed is ok. T0 have 0° temperature.

miblooming, 2017-05-21:
Please change tools count to 2.
[image]

DaGov, 2017-05-21:
[image]

DaGov, 2017-05-21:
I changed.

miblooming, 2017-05-29:
Maybe you should also set tools count to TWO in marlin。

miblooming, 2017-05-29:
If you connect to PC with "Repetier-Host" or "pronterface.exe"， does the  temperature T0 display? If not display, you should set tools count to TWO in marlin frst。

DaGov, 2017-05-29:
I use a smoothieware. And if I connect to PC I use 2 hotends. This is your bug.

DaGov, 2017-05-29:
I connect with Simplify3D. And T1 and T0 display temperature.

miblooming, 2017-05-29:
Please input M105 In Gcode, and tell me the output.
[image]
[image]

DaGov, 2017-05-29:
[image]

DaGov, 2017-05-29:
I changed designator T to T0 and T1 to T1.
[image]
but I have new trouble
[image]

DaGov, 2017-05-29:
If I back to T and T1 temperature T1 and bed correct. T0 is 0.

miblooming, 2017-05-30:
It's a bug of compatibility with "smoothieware". I'll fix it.Thank you.