miblooming/JZ-TS24-2 #4: adding gcode to button tiles

mikemayhak, 2017-11-25:
hello is there a way to program gcode to the icons or blocks tried to move home xyz to home screen but blocks will not work.
Also is there an option to change back ground to black in jzhelper?

miblooming, 2017-12-10:
Please translate the helper file into you language by google translate, there may be some useful infomation.
Q1: 
STEP1:Add a gcode "G28 X0 Y0 Z0“
STEP2:Add the gcode id to UI Element‘s Action attribute
STEP3:Set the TCH attribute of UI Element Checked

Q2:
Back ground is unchangeable now
But a block of black of size 320*240 can cover the white backgroud to black
I will add a feature to change back grould color later.

Q3:
Wifi is still under developing

miblooming, 2019-05-15:
WIFI module
https://item.taobao.com/item.htm?id=581913421214
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5)

closed 2019-05-15 by miblooming