miblooming/JZ-TS24-2 #5: Smart display for CNC milling machine

zagranichnyy, 2017-12-25:
when we wait for the firmware to work the display together with the CNC milling machine)))?

miblooming, 2018-01-02:
@zagranichnyy It's possible to work with CNC machine in theory. What motherboard and firmwave does your CNC using?

zagranichnyy, 2018-01-02:
in the firmware repetier There is an opportunity to work with milling and laser cnc. marlin with марлин с неб марлин с марлин с марлин с марлин с марлин с марлин с марлин с небольшими

zagranichnyy, 2018-01-02:
[image]

miblooming, 2018-01-03:
Good work. If you need any help on trying，please send me message，I'll try my best to afford assistance.

zagranichnyy, 2018-01-03:
leave your email address or email me v.zagranichnyy@gmail.com  I think that together it will be easier)