miblooming/JZ-TS24-2 #7: Wifi on JZ-TS35

cabped, 2019-03-08:
On the JZ-TS35 there is a connector for a WiFi extension module. I’ve searched the web but not found the specific module that goes with JZ-TS35. I was wondering if it would work with ESP-3D module (https://github.com/luc-github/ESP3D)? Case not, once ESP-3D also uses the Aux-1 in RAMPS there’s a alternative way to connect JZ-TS35? And also: which is the WiFi extension module compatible anyway?

miblooming, 2019-03-08:
WIFI module：https://item.taobao.com/item.htm?id=581913421214
WIFIHelper for PC：https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5)
This module is based on ESP32 with a custom firmwave for JZ-TS35.

cabped, 2019-03-09:
Thanks for the answer. Is it sold on eBay or AliExpress or other site that ships to outside China?

cabped, 2019-05-03:
Got the module today. Works preatty fine. Fast connection for file transmition, much better than other solutions. However, if I may ask, I would request, if possible, implementing on the firmware a web server for access from other OS as Mac, as avaiable in ESP-3D module (https://github.com/luc-github/ESP3D). That would make JZ=TS35 cross platform on connections, and also allow remote monitoring of the printing process from outside local network. It would be very usefull also to add an open port to connect to the printer as a serial TCP/IP hub from Repetier Host, for example.

miblooming, 2019-05-05:
Thinks for your feedback. 
Good ideas, I'll try to develop those features.
And a cura plug for  JZ-TS35 WIFI is under developing.

cabped, 2019-05-05:
Perfect. I’m looking forward to it. Thanks again.

miblooming, 2019-05-15:
Cura plugin for WIFI in released. Follow:https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5)

closed 2019-05-15 by miblooming

TAOGde, 2019-06-18:
> WIFI module：https://item.taobao.com/item.htm?id=581913421214
> WIFIHelper for PC：https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5)
> This module is based on ESP32 with a custom firmwave for JZ-TS35.

Where can I found the custom firmware? your firmware version inside your https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5) folder does not update for my ghost4 latest version. (board hardware version is 1.4)

miblooming, 2019-06-19:
Here https://github.com/miblooming/JZ-TS24-2/tree/master/other/flyingbear

TAOGde, 2019-06-19:
> Here https://github.com/miblooming/JZ-TS24-2/tree/master/other/flyingbear

GREAT work, it works :)

Can I send you the German translation for your firmware? I am nativ speaking German and something looks wired to me in your translation.

than can I change with your tool for my ghost4 the gcode group commands?

regards Torsten

miblooming, 2019-06-24:
1.translation
Yes, please edit this file for translation, and send back to me.

[language-Deutsch.zip](https://github.com/miblooming/JZ-TS24-2/files/3319868/language-Deutsch.zip)

2.gcode group
The firmwave of ghost4 is a custom version. Please use the UIEditor here:
https://github.com/miblooming/JZ-TS24-2/tree/master/other/FlyingBear_TS_WIFI/UIEditor

rustam-rezvanov, 2019-10-16:
> 
> […](#)

Hi! Did you manage to find this module on AliExpress ?