miblooming/JZ-TS24-2 #11: Bad positioning back Z when pause, power off or end-of-filament

cabped, 2019-06-09:
Hello,

I’ve just updated my JZ-TS35 firmware to the latest version 3.6.2, but I still got an issue with the position of the tool, when it resumes the print:

(A) When I press pause button on screen, the bed will move down some distance and the tool is set to home position, print pause. When I press play, the bed goes up a little, but not enough to reach where it stopped. The X and Y seems to be set correctly, but not the Z.

(B) When I turn off during the print, when I turn it back on, and try to print a file, I receive I message of recovery. Then, I try to resume the print. The bed and tool are heated up. The bed moves down a bit, the tool is set to home, and back to where it stopped. Then the Z is set to the position where it stopped (in my test it stopped at 1.20 and was set back to 1.20), but the tool wasn’t quite where it stopped (was I little higher) so the resume didn’t worked well. I’m supposing it didn’t consider the offset??

(C) When the filament sensor detects a end of filament, the Z moves down, and X Y are set home. When the filament is filled up, and I press play, the X and Y are correctly set back where the tool stopped, but not the Z so the bed stands lower than where it should be to continue the print. (I’ve 
made a video which is attached to this).

Thanks.

cabped, 2019-06-09:
This first video is when it pause because of the end of filament.
[Pausing.zip](https://github.com/miblooming/JZ-TS24-2/files/3269996/Pausing.zip)

cabped, 2019-06-09:
This video shows the bad positioning of Z when it resumes from end of filament.
[Resume.zip](https://github.com/miblooming/JZ-TS24-2/files/3269999/Resume.zip)

miblooming, 2019-06-19:
Yes it didn't consider the offset of Z. 
What's the offset value of the Z. 
Please try  to  set the offset of Z to zero, and check whether the problem is gone.
Maybe there is a bug.

cabped, 2019-06-20:
The offset value of the Z axis is set to fine adjust the distance from the nozzle bottom to bed (and the top of the object in the other layers) on a 3D printer with automatic bed leveling as mine.

It can’t just be set to 0 because the result would be a bad first layer (too squeezed).

The value depends on the nozzle position (changes whenever the nozzle is changed) and now it’s about 2.5mm.

It’s set due printer calibration.

But the issue doesn’t seems to be just the offset miss. In most of the cases related on mi request, the distance from the nozzle to the object is much over 2.5mm (seems to be over 10mm).

Perhaps the problem on the code is the variation on considering the absolute/relative position of the Z axis??

There most be a way to register on pause the absolute position of Z and call it back when resume printing, because seem the code assumes the final position based on the moves the tool makes to get out from the printing position when a pause is called (so it’s based on relative positioning of Z).

Maybe the current code works successfully on static printing head printers (which the bed moves on all axis) but not on moving heads like mine, on which the bed only moves on the Z axis??

The fact is: on the current version, printing pause doesn’t work for my printer.

If you wish, I can send you detailed videos of the process that aren’t working.

Best regards,

Enviado do meu iPhone

miblooming, 2019-06-24:
In firmware of  version 3.6.2.It should work fine in condition(A) and (C). 
When pause by hand, the value offset of Z is not lose, and the Z original position is not lose too.
When resume by hand, screen just tell the printer move tool to the position, which saved on pause.
It's a strange bug.

The distance of Z in condition(B) should equal to the offset of Z, because the screen lose the info of Z offset, which calculate by automatic bed leveling module

Can you send me your marlin, gcode file and printing.sys file.
marlin: there are many version of marlin
gcode file: on your video
printing.sys: When printing task is paused, there's a printing.sys file in SD card.

cabped, 2019-06-25:
Well, I don't know if it's relevant, but I did another test: I tried to force Z up in condition (A), by manually positioning it with the move axis option on the panel, while the print was paused. The bed came up (Z up) a little (the command was to lift it 10mm, not sure if it did it all), but not enoght to set it on the top of objetct. At some point I couldn't lift it up more, as the screen informed my Z axis was in 0 position. I didn't understand that... (?) How it is in Z=0 if the print was paused while doing the layer 3, about 1.2mm over the bed...

My Marlin version is 1.0.6. I'll provide the files requested as soon as possible.

miblooming, 2019-06-26:
I know too factor about z-offset. 
1.UIEditor of JZ-TS. Lift Z before pause
2.Marlin config. Lift Z before homing XY
[image]
[image]

cabped, 2019-06-26:
Which GCode is sent to set the tool on home position when manually we pause print? Doesn’t it loses the Z reference on doing that?

Which GCode is sent to pause? I mean: does the screen send a print command to Marlin or it keeps managing the process by sending  sliced file gcode line by line and stops on pause hit?

Perhaps setting the X Y and Z by movement Gcode commands instead of using home command would work?

The “Lift Z on Pause(mm)” parameter on the screen setting is added to the “Z_RAISE_BEFORE_HOME”? (For example, in the settings below the home will set Z to 9?)

I’ll test by setting “Lift Z on Pause(mm)” to 0 and disabling the Home X Y on Pause...

How does the “Move to Point X” works over the Home X Y on Pause enabled?

Thanks

Enviado do meu iPhone

miblooming, 2019-06-27:
On printing, JZ-TS send gcode cmd line by line in Gcode File.
On pause
STEP 1 [stop read gcode cmd from gcode file]
SETP2 [save printing state]
SETP3 send these gcode cmd:(<expression> will be cal calculated to number)

**[if Lift Z on Pause(mm) > 0] G1 Z<current_z + Lift Z on Pause(mm)>** 
[if Home X&Y on Pause  checked] G28 X0 Y0 
**[else] G1 X<Move to Point X on Pause> Y<Move to Point Y on Pause>**

On resume
[if Home X&Y on Pause  checked] G28 X0 Y0 
G90
G1 Z<z on paused + 3>
G1 X<x on paused> Y<y on pause>
G1 Z<z on paused>
G92 E<e on paused>
[if is on relative mode when paused] G91 
[else] G90
G1 F<f on paused>

JZ-TS almost does noting dubious when clicking Pause on screen.

miblooming, 2019-07-22:
Has the problem been solved?

miblooming, 2019-07-22:
Has the problem been solved? @cabped

cabped, 2019-07-23:
Not yet. I’m outside town so I couldn’t send to you the files requested. I’ll do it as soon as I get back, ok?

Enviado do meu iPhone

cabped, 2019-07-23:
PS: I haven’t updated the screen firmware to the las version recently released. Was there any changes to solve my issue? If so, I’ll update it as soon as I get back and do some testing.

Enviado do meu iPhone

miblooming, 2020-02-02:
Please try it, the default setting of pause is changed to Dock, not Homing.