miblooming/JZ-TS24-2 #16: Backlight control.

DrOnJD, 2019-07-06:
Hey. How can I add a button that will control the backlighting of my printer?
That is, on the first click, execute the command “M355 S1 P255”, and on the next click execute “M355 S0 P255"?

miblooming, 2019-07-07:
Use uieditor, please try it.

JZ-TS35 https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5)/UIEditor

JZ-TS24
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS24-3____(2.4%262.8)

DrOnJD, 2019-07-07:
I watched this program before writing. There it is not possible to do this if I’ve done it right. Only one gcode can be assigned to one element, that is, either turning the backlight on or off. Is it possible to do something on one button in turn, either on or off?

miblooming, 2019-07-08:
Yes, the editor does not support Switch Button. I'll add it to my develop list.

DrOnJD, 2019-07-08:
Thank you very much.

miblooming, 2019-07-22:
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)

[image]

closed 2019-07-22 by miblooming