miblooming/JZ-TS24-2 #17: JZ-TS35WIFI__(3.5)_portrait mode

edsant1812, 2019-07-18:
I'm launching a new line of 3D printers in my country and I'm using the JZ-TS35WIFI__(3.5) diaplays. I wonder if it would be possible to use them in portrait mode, similar to the JZ-TS24

miblooming, 2019-07-18:
Yes. Please give me some infoamtion of your company.
We can make a portrait theme for you.

edsant1812, 2019-07-18:
perfect, what information is needed and in which email I can pass this information

miblooming, 2019-07-18:
My email is loomingxu@qq.com.
Can you send us the introduction of you company and product.

miblooming, 2019-07-22:
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/firmwave%20lastest_portrait
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/UIEditor__portrait
[image]

closed 2019-07-22 by miblooming