miblooming/JZ-TS24-2 #18: New menue for IDEX or DUAL_X_CARRIAGE Printer

PI-Beginner, 2019-08-11:
Can you create an Menue for IDEX Printers? Marlin call: DUAL_X_CARRIAGE

Menue like Pictures from standard LCD 

It should also shows the different IEDEX modes: Copymode 

That would be a unique point for this display, nobody else has. 

[image]
[image]
[image]

miblooming, 2019-08-11:
I'll try it.

miblooming, 2020-02-03:
Please check out the new version of V3.8.2
IDEX menu added.
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)

closed 2020-02-03 by miblooming