miblooming/JZ-TS24-2 #19: Endstop monitoring triggers BLTouch to deploy probe

EtherToroid, 2019-09-06:
Hello. I've run into a problem, using JZ-TS40 display from FlyingBear and skr 1.3 mb. It has been configured with the v.2.4.2.2 JZ Ui helper. My BLTouch probe always deploys the probe, even when the printing started, causing crashes. The blue light in it flashes with a period of a second. My experiments using regular display (Reprap full graphics) did'd show this problem. Thus I assume, that the endstop monitoring feature of the display triggers the probing and thus causing the effect. I would appreciate any solution, such as an option to stop endstop monitoring proccess or somehow block the M119 command.

miblooming, 2019-09-07:
Please try the new firmware: https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/firmwave%20lastest_landscape

EtherToroid, 2019-09-07:
Unfortunately, this version hangs at 100% Update Firmware status. For at least 15 minutes now.

miblooming, 2019-09-07:
Delete jzwifi.bin before upgrade. This firmware is not compatible with the custom firmware of FlyingBear on WIFI

EtherToroid, 2019-09-07:
Still doesn't work. Tried multiple times and still hangs. Whereas my configured firmware flashes fine.

miblooming, 2019-09-07:
Please download the new firmware, and try again. There's a complie error.

EtherToroid, 2019-09-07:
I've managed to flash the newer version and now problem is no more. But if I go to the endstop monitor page, it still triggers the bltouch to deploy. Thank you for solving the issue, look forward to be able to customize this version.

closed 2020-01-22 by EtherToroid