miblooming/JZ-TS24-2 #20: wifi module

alexxxdev, 2019-09-09:
Hello
I want to install the latest firmware on ghost4, since the firmware from Flyingbear Ghost 4 is 3.6.5,
3.7.2 firmware is installed, it works, but wifi does not work
As I understand it, another wifi module is used on Flyingbear Ghost 4, can we expect its support?

miblooming, 2019-09-10:
The touchscreen for Flyingbear Ghost 4 is a custom version of JS-TS40. It's custom for  Flyingbear Ghost. There wifi module is not compatible with the public version.If you use the touchscreen with Flyingbear Ghost, please use the custom version. It's more suitable. If you want to use it out of Flyingbear Ghost, the public version of JZ-TS will be batter.

miblooming, 2020-02-02:
I'm sorry that it is not open source.

closed 2020-02-02 by miblooming