miblooming/JZ-TS24-2 #21: JZ-TS40 BabbyStepping

ffnull, 2019-09-23:
Hello.Recently buy your display. Best controller ever. Wifi printing is awsome, but i need babbystepping to correct first layer. How i can add it or can you add it?

ffnull, 2019-09-28:
Project abadoned? no support at all?

miblooming, 2019-09-28:
Sorry, I am not  familiar with babbystepping. I am learning about this iusse.

ffnull, 2019-10-06:
So? Any solution? http://marlinfw.org/docs/gcode/M290.html

miblooming, 2019-10-06:
Three methods：
1.Add some codes in Gcode file by modifing the setting "Starting G-Code" of slicer software
2.Add a button on screen by UIEditor to trigger G290-operation  by handwork
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)
https://www.youtube.com/playlist?list=PLamn4I4Ptg5KcN8KiLsWBeAiS-ABRkXjR
3.Add an event operation of   by UIEditor 
[image]

ffnull, 2019-10-06:
I need add it to main screen of printing process. Button must be near buttons of corrections of temperatures of heaters. on center of button must be shown real height of Z and it must be change while clicking up and down

ffnull, 2019-10-06:
Here another example how its work https://youtu.be/s5kld8MIo6M

miblooming, 2019-10-06:
Please try it on UIEditor.
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)
https://www.youtube.com/playlist?list=PLamn4I4Ptg5KcN8KiLsWBeAiS-ABRkXjR

ffnull, 2019-10-06:
by the way. no sizes and no models for jz-ts40.

miblooming, 2019-10-06:
The firmware of jz-ts40 is same as jz-ts35

RV3Dtech, 2019-11-13:
[image]

[image]

miblooming, 2019-11-14:
> by the way. no sizes and no models for jz-ts40.
Here:
https://github.com/miblooming/JZ-TS24-2/blob/master/size_shell/JZ-TS40_4.0shellV1.0-solidworks2016.zip

closed 2020-01-22 by miblooming