miblooming/JZ-TS24-2 #25: API documentation

tuzkov, 2019-10-30:
Hello!
Do you have some kind of documentation for screen API?
By Cura plugin I find some APIs, but I not sure that is all.
At least I want to switch screen to "offline" mode by WIFI to print via USB.
Thank you!

miblooming, 2019-10-30:
There's no code to set offline. I can add some api and give you a document at next version. Maybe somedays later.

tuzkov, 2019-10-30:
Nice! Thank you!

miblooming, 2019-11-06:
Please use the new firmware
V3.7.6
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/firmwave%20lastest_landscape

API file here:
https://github.com/miblooming/JZ-TS24-2/blob/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/API_for_wifi.txt

Send 7998 Disconnect
Send 7993 Reconnect

Zorchz, 2019-11-11:
Is it possible to update flyingbear version with that change?
I'm about https://github.com/miblooming/JZ-TS24-2/tree/master/other/flyingbear/firmwave

miblooming, 2019-11-14:
I'm sorry. The flyingbear version is a custom version. I'll not upgrade it privately. 

 
&mdash;

closed 2020-01-27 by miblooming