miblooming/JZ-TS24-2 #28: Problem with Leveling with JZ-TS28 and others question

Sharkys80, 2020-01-14:
Hi

I have a Problem with Leveling with  JZ-TS28 , my 3D touch dosnt work with the Display.
Without the Display its works fine. After the first Point the Sensor don't deploy.
Is the Leveling Option on Display not G28 &G29??

No language support in German for the 2.8 only for 3.5?

Any Code for botton Sends i dont find the correct code in the list i try build Gode M122 and i need the feedback from the TMC ?

And the last question :any other option to buy a 3,5 or 4 display from taobao  site?

miblooming, 2020-01-15:
Hi
1. 3D touch
The leveling of jz-ts28 is manually. Moving to points, waiting you to adjust the screws of bed.
If you want to use g28&amp;g29, please add a button to send gcode of g28&amp;g29 by UIEditor.
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS24-3____(2.4%262.8)/JZ-TS24-3____(2.4%262.8)_landscape

2.Yes there's no German for the 2.8

3.Two method:
A) Send gcode by Gcode Common Line, HomePage -&gt; My Tools -&gt; Gcode
B)Add buttons to send gcode by UIEditor.

4.Pay me by paypal, and I'll ship the product to you.

Best Regards
LoomingXu

 
 Is the Leveling Option on Display not G28 &amp;G29??

miblooming, 2020-01-15:
It's for JZ-TS35/40, some codes are not enable on jz-ts28
 
https://github.com/miblooming/JZ-TS24-2/blob/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/API_for_wifi.txt

Sharkys80, 2020-01-15:
Thank you for the answers.
But yesterday a start a test with only printing from SD:
So my normal config is UBL is on and setup is complet in Marlin+Simplify3D(start script) + Eprom on LPC1768 but when i start.He ignored the the start script wit G28 & G29 J in the gode, the Z axis  dosnt compensate its self. printig without Display with only  a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER its work fine .What is the failure?

miblooming, 2020-01-15:
Can you show me the command lines at beginning of Gcode file?

Sharkys80, 2020-01-15:
G-Code generated by Simplify3D(R) Version 4.1.2
; Jan 15, 2020 at 9:15:00 AM
; Settings Summary
;   processName,Process1
;   applyToModels,geh?use
;   profileName,Default (modified)
;   profileVersion,2017-03-01 08:00:00
;   baseProfile,Default (modified)
;   printMaterial,PLA
;   printQuality,Fast
;   printExtruders,
;   extruderName,Primary Extruder
;   extruderToolheadNumber,0
;   extruderDiameter,0.4
;   extruderAutoWidth,1
;   extruderWidth,0.48
;   extrusionMultiplier,0.9
;   extruderUseRetract,0
;   extruderRetractionDistance,1
;   extruderExtraRestartDistance,0.6
;   extruderRetractionZLift,0.6
;   extruderRetractionSpeed,460
;   extruderUseCoasting,0
;   extruderCoastingDistance,1
;   extruderUseWipe,0
;   extruderWipeDistance,5
;   primaryExtruder,0
;   layerHeight,0.3
;   topSolidLayers,5
;   bottomSolidLayers,5
;   perimeterOutlines,5
;   printPerimetersInsideOut,0
;   startPointOption,2
;   startPointOriginX,0
;   startPointOriginY,0
;   sequentialIslands,0
;   spiralVaseMode,0
;   firstLayerHeightPercentage,80
;   firstLayerWidthPercentage,80
;   firstLayerUnderspeed,0.6
;   useRaft,0
;   raftExtruder,0
;   raftTopLayers,3
;   raftBaseLayers,2
;   raftOffset,3
;   raftSeparationDistance,0.14
;   raftTopInfill,100
;   aboveRaftSpeedMultiplier,0.3
;   useSkirt,0
;   skirtExtruder,0
;   skirtLayers,1
;   skirtOutlines,5
;   skirtOffset,0
;   usePrimePillar,0
;   primePillarExtruder,999
;   primePillarWidth,12
;   primePillarLocation,7
;   primePillarSpeedMultiplier,1
;   useOozeShield,0
;   oozeShieldExtruder,999
;   oozeShieldOffset,2
;   oozeShieldOutlines,1
;   oozeShieldSidewallShape,1
;   oozeShieldSidewallAngle,30
;   oozeShieldSpeedMultiplier,1
;   infillExtruder,0
;   internalInfillPattern,Fast Honeycomb
;   externalInfillPattern,Rectilinear
;   infillPercentage,35
;   outlineOverlapPercentage,15
[... 114 lines cut ...]
;   toolChangeRetractionSpeed,600
;   externalThinWallType,0
;   internalThinWallType,2
;   thinWallAllowedOverlapPercentage,10
;   singleExtrusionMinLength,1
;   singleExtrusionMinPrintingWidthPercentage,20
;   singleExtrusionMaxPrintingWidthPercentage,200
;   singleExtrusionEndpointExtension,0.2
;   horizontalSizeCompensation,0
G90
M83
M106 S128
M140 S65
M190 S65
M104 S235 T0
M109 S235 T0
G28 ; Alle Achsen Nullen
G29 L0 ; L0 =ein punkt J= drei Punkte
; process Process1
; layer 1, Z = 0.240
T0
; feature outer perimeter
; tool H0.240 W0.384
G1 X0.000 Y0.000 Z0.240 F4600
G1 X125.977 Y120.784 Z0.240 F4600

miblooming, 2020-01-15:
Maybe there's a compatibility issue with 3D touch.
I'll try to fix it. But no that quickly.

Sharkys80, 2020-01-15:
OK i can wait thank you, but i will test a Inductive sensor on my second printer. I guess that dosent work too.

Sharkys80, 2020-01-22:
So i tested with my 2 printer Melziboard 2.0 and Inductive sensor: You have right, there's a compatibility issue with 3D touch the other Sensor work fine with ABL and display. I hope for fix