miblooming/JZ-TS24-2 #29: tft38

vzagranichnyy, 2020-01-19:
Hi, I have jz-tsx-v1.4 tft 38. it has been idle for a long time, the stm32 processor burned out in it and I ordered a new processor, the wizard re-soldered it, but after turning on the white screen is on and it does not flash, I think you need a bootloader for it, could you help me with this?

miblooming, 2020-01-19:
I'm sorry that bootloader is not open.

vzagranichnyy, 2020-01-19:
> Извините, что загрузчик не открыт.

can I flash it?

miblooming, 2020-01-19:
No. I have some old  jz-tsx board. jz-tsx is suspend production. I can give it to you as a gift, if you pay the shipping fee.

vzagranichnyy, 2020-01-19:
> У меня есть старая доска JZ-TSX. JZ-TSX приостановить производство. Я могу подарить его вам, если вы оплатите стоимость доставки.

This is very generous.  I have a friend's address in Beijing.  Tell me how much to pay.  I can use web money or a visa card.

miblooming, 2020-01-19:
To Beijing, shipping to you friend. And your friend  pay the shipping fee. Is it ok?

vzagranichnyy, 2020-01-19:
> В Пекин, доставка вашему другу. И ваш друг оплатит стоимость доставки. Это нормально?

I will pay for the delivery myself.give your details.

miblooming, 2020-01-20:
Connect next year. We are all on holiday for Chinese New Year before 2020-1-30.

Best Regards
LoomingXu