miblooming/JZ-TS24-2 #31: There is no way to change the print head auto-parking code

Pilson66, 2020-01-28:
I found that there is no way to change the print head auto-parking code.
I tried to do this this in "GCODE" page, GcodeID = 8031, and I changed the line to mine (I need this workaround for some reason):
Before:
G28 X0 Y0 Z0
After:
G28 X0
G28 Y0
G28 Z0
And after assembling the firmware, I found out that this code is not being called. Instead, the old code is called.
After that, Ive go to page "MOTION", and found that the block of movements along the axes, and the auto-home button do not have their own separate elements, and are made as one large picture, without the possibility of modifying each of these elements separately.
Can this bug be fixed? Or at least add out the GcodeID, in which can edit the Gcode for  autohome?

Also, where can I edit the "Pause/Stop/Resume" Gcode ? It also need for some reason, related to the sad "feature" of my printer. (It need to disable stealthchop during print, and enable stealthchop at pause/stop/resume, before the display sends an auto-home command to the printer.).

miblooming, 2020-02-02:
1.MOTION
The order of Home of XYZ can be set in marlin.
https://github.com/MarlinFirmware/Marlin/blob/e6a37ba2a05131873b4b1d1f2b6b2697c388f944/Marlin/Configuration_adv.h
LINE 593: //#define HOME_Y_BEFORE_X

2.MOTION "Pause/Stop/Resume"
Please try it on UIEditor: UI -&gt; Custom Button -&gt; Event Trigger

Best Regards
LoomingXu

miblooming, 2020-02-02:
[image]

Pilson66, 2020-02-02:
What exactly needs to be entered in the "Action" field? I tried to enter the Gcode string and GcodeID, in both cases it reporting about syntax error.
[image]
[image]

miblooming, 2020-02-03:
[image]