miblooming/JZ-TS24-2 #32: Please update russian translation.

Pilson66, 2020-02-02:
I am a native speaker of the russian language, and I tried to translate the entire interface as correctly as possible.
Please update the translation. The file with the translation is attached.
[jz-ts_russian_translation.xlsx](https://github.com/miblooming/JZ-TS24-2/files/4144640/jz-ts_russian_translation.xlsx)

Also, the program does not have the ability to change ID 2021. There need to change "Русский язык" to "Русский".
Thanks.

miblooming, 2020-02-03:
Please check it here.
New russian translation.
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)

closed 2020-02-03 by miblooming

Pilson66, 2020-02-03:
Reviewed again in detail, and found a few more inaccuracies. Please update.
File is attached.
[jz-ts_russian_translation_update_1.xlsx](https://github.com/miblooming/JZ-TS24-2/files/4147672/jz-ts_russian_translation_update_1.xlsx)

miblooming, 2020-02-03:
Done.
Please check it.

Pilson66, 2020-02-03:
It looks quite presentable.
Thanks.

Pilson66, 2020-02-03:
One more (maybe last) correction, please update.
[jz-ts_russian_translation_update_2.xlsx](https://github.com/miblooming/JZ-TS24-2/files/4148650/jz-ts_russian_translation_update_2.xlsx)

miblooming, 2020-02-05:
It's done.

Pilson66, 2020-02-10:
Please update one more translation.

ID|China|English|Russian
----|-----|-----|-----
1163	|	数值过大	|	Too large	|	Значение слишком большое

miblooming, 2020-02-11:
OK. Next update.

Pilson66, 2020-03-05:
Hello. Any updates?

miblooming, 2020-03-06:
1136 is done.