miblooming/JZ-TS24-2 #33: Cura plugin failure when trying to send file

Pilson66, 2020-02-03:
Today I updated the plugin for Cura 4.4.1 Linux.
Several problems at once:

Case 1. The plugin is not visible until replace the line:
[image]

Case 2. When go to the "ZJTSWIFI" plug-in tab, the Cura crashes until I comment out the following lines. And atfer that, I need to manually add IP address of my printer to JZ_TS_WIFIPlugin/res/MyMain.qml.
[image]

Case 3. When I trying to send a file to the printer, the Cura crashes aiagin.

Screenshots:

For case 2:
`Traceback (most recent call last):
  File "/home/pilson/.local/share/cura/4.4/plugins/JZ_TS_WIFIPlugin/JzTsWifiStage.py", line 62, in init_jz_plugin
    self.plugin_main.init_stage(self.plugin_ui_item,self.upload_file_name)
  File "/home/pilson/.local/share/cura/4.4/plugins/JZ_TS_WIFIPlugin/PluginMain.py", line 365, in init_stage
    self.bind_ui()
  File "/home/pilson/.local/share/cura/4.4/plugins/JZ_TS_WIFIPlugin/PluginMain.py", line 405, in bind_ui
    self.update_history_ip()
  File "/home/pilson/.local/share/cura/4.4/plugins/JZ_TS_WIFIPlugin/PluginMain.py", line 313, in update_history_ip
    model.setItemData(model.createIndex(i,0), {1:self.ip_list[i],0:"True"})
RuntimeError: no access to protected functions or signals for objects not created from Python`

`Thread 0x00007fbb367fc700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbb36ffd700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 549 in wait
  File "/tmp/.mount_curaB1CSdD/usr/bin/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbb377fe700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbb37fff700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbb54bee700 (most recent call first):
  File "/tmp/.mount_curaB1CSdD/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbb553ef700 (most recent call first):
  File "/tmp/.mount_curaB1CSdD/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fbb74ff9700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
[... 59 lines cut ...]
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 306 in _logInfoWidget
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 156 in _createDialog
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 77 in __init__
  File "/build/4.4/build/inst/bin/cura", line 111 in exceptHook
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 341 in exec_
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 804 in run
  File "/build/4.4/build/inst/bin/cura", line 180 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
`

For Case 3:
[image]

miblooming, 2020-02-03:
I only test it on Windows.
So there maybe some bugs on Mac and Linux.
I'll fix it Later.

Super-Matter, 2020-02-09:
Have the same problem on Windows 10.

miblooming, 2020-02-10:
Please use WifiHelper temporarily.
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/WifiHelper

I'll fix the cura plugin.

miblooming, 2020-03-15:
not yet