miblooming/JZ-TS24-2 #35: Blank white screen after splash screen

Mayoogh, 2020-04-30:
Blank white screen after flashing the `UPDATE3.BIN` and `custom.bin` created using **JZHelper V2.3.2.0** . No issue with [this](https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS24-3____(2.4%262.8)/firmwave%E6%9C%80%E6%96%B0%E5%9B%BA%E4%BB%B6/landscape%E6%A8%AA%E5%B1%8F) latest firmware. The main reason I am editing the firmware is that the print is resuming after lifting 5mm from the original position. 

## Expected Behavior

After splash screen (logo) the `HOME` should be loaded

## Current Behavior

After splash screen (logo) stuck with a white screen without loading the `HOME` screen 

## Steps to Reproduce

1. Create custom firmware using JZHelper V2.3.2.0 `JZ-TS24-3____(2.4&2.8)` -> `JZ-TS24-3____(2.4&2.8)_landscape` -> `JZHelper.exe`
2. Update it on 2.4 size display.

## Environment

- Windows 10, 64bit
- JZHelper V2.3.2.0

Sacred-Shadow, 2020-04-30:
Why UPDATE3.BIN? 

The JZHelper creates an UPDATE2.BIN and a custom.bin for you.
If recognized, both are also written to the USB stick / SD card.

Mayoogh, 2020-04-30:
[image]
I am getting `UPDATE3.BIN` and `custom.bin` files as output when the build is completed.
Only `UPDATE3.BIN` is getting written to SD card.