miblooming/JZ-TS24-2 #43: how to increase the transfer speed over Wi-Fi

dacentichka, 2020-11-23:
good day skadite how to increase the speed of transmission over wifi

dacentichka, 2020-11-23:
[image]

miblooming, 2020-11-23:
1. Use a faster SD card, recommend a card of class 10 or higher
2. Make the printer in a position with strong WIFI signal
3. Reduce the number of files stored on the SD card

dacentichka, 2020-11-23:
and what is the maximum speed can be?, please attach a photo, and if it is not difficult for you, tell us what interface Wi-Fi records files?

miblooming, 2020-11-24:
about 200KB/s. The ability of the STM32 Chip is not that powerful to receive data from wifi.
Try this:
https://github.com/miblooming/JZ-TS24-2/tree/master/JZHelper_ui_editor/JZ-TS35WIFI___(3.5%264.0)/WifiHelper
Is same as that cura plugin.

dacentichka, 2020-11-30:
thank you, this did not help, I will continue to use a third-party plugin for Kura, as it is more functional. have a nice day)

closed 2020-11-30 by dacentichka